adafruit/Adafruit_MCP4725 #6: Compiler Errors when Using Arduino Due (Uno works ok)

bpsengineer, 2015-11-06:
With Arduino Due selected in IDE 1.6.5, when I try to compile the examples, or even an empty file with only Wire.h and this library included, I receive the following errors:
C:\Users\Brian\Documents\Arduino\libraries\Adafruit_MCP4725\Adafruit_MCP4725.cpp: In member function 'void Adafruit_MCP4725::setVoltage(uint16_t, bool)':
C:\Users\Brian\Documents\Arduino\libraries\Adafruit_MCP4725\Adafruit_MCP4725.cpp:67:22: error: 'TWBR' was not declared in this scope
uint8_t twbrback = TWBR;
^
Error compiling.

If I switch over to Arduino Uno, I receive no errors and my program runs.
Is there a different MCP4725 library for Due?  How do I fix this? I downloaded the library on 10/30/2015 and am using IDE 1.6.5

pwspania, 2017-07-13:
It doesn't look like this library supports the Due. It also hard codes using `Wire`, with no configuration options to use `Wire`.

ladyada, 2020-06-01:
Updated latest version will allow selecting any TwoWire

closed 2020-06-01 by ladyada