adafruit/Adafruit_MCP4725 #7: Generate 2.55 Volt DC (stable)

Mondeo, 2016-03-15:
Hello,
I like to generate a stable output of 2.5Volt DC.
Maybe someone can help me to find a solution.
Thank you in advance, Urs

Cornelis85, 2016-05-14:
The MCP does not have an internal reference. The output is allways a fraction of the supply voltage Vdd.
For accurate applications, you have to connect the Vdd-pin of the breakout board to a voltage reference device (for instance REF02).

mitchyboy9, 2017-03-08:
In case it helps anyone with a similar issue, my version of this library allows you to set a voltage level directly, rather than having to calculate the input code that corresponds to a particular voltage.  As Cornelis85 correctly says, you will still need a stable reference voltage, but at least you don't need to work out the input code, which will probably make your sketch clearer.

My fork of this library:
[https://github.com/mitchyboy9/Adafruit_MCP4725](https://github.com/mitchyboy9/Adafruit_MCP4725)

Example usage to make the MCP4725 generate a 2.55V output, given a reference voltage of 5V:
`setNearestActualVoltage(2550, 5000, false);`
Where `2550` is 2.55V expressed in mV
`5000` is the 5V reference voltage in use, expressed in mV
`false` tells the MCP4725 _not_ to store this output level in EEPROM

This issue against the library should probably be closed now.

kbowen99, 2017-07-31:
I have no Idea how accurate the readings are, but would it be practical to implement the code given [here](https://hackingmajenkoblog.wordpress.com/2016/02/01/making-accurate-adc-readings-on-the-arduino/) to get the reading at the rail and pass that input into your setNearestActualVoltage() function? I don't have a trusted reference (nothing more than a multimeter), but it would be pretty cool for the library to support setting actual voltages.

mitchyboy9, 2017-08-01:
Yes, that should work nicely.

I tried the code in that post a little while back and was pleased with the results, although I did need to calibrate the 1125200 value for my specific Arduino as it was slightly off (as mentioned at the bottom of the post).  Seeing as you have a voltmeter, you will be able to do this, but remember that you need to do it for every individual chip e.g. if you swap one Arduino for another, you will most likely need to edit that value in the sketch.

agaelema, 2017-08-01:
I do not believe that use the arduino to measure rail voltage will be the best thing, remember that the resolution of most Arduinos is 10bit, 4 times lower than resolution of MCP4725. The error in the measurement without some calibration can be greater than the error of a external reference.

As @Cornelis85 said MCP4725 does not have an internal reference, but use the power rail as reference. You can use an external reference of 5V as REF02 or REF5050 (better), or even REF5040 of 4.096V, reaching theorical 1mV/bit. Just remember to not overload this references powering other devices. Overload will change the voltage value, decreasing the accuracy.

ladyada, 2020-06-01:
resolved

closed 2020-06-01 by ladyada