adafruit/Adafruit_MCP4725 #8: Two MCP4725

Cornelis85, 2016-05-14:
For my project I have t use two ADC's.
The MCP4725 has a pin A0  whichs allows to use two ADC's, 

The problem is that the I2C-address has been fixed programmed wqithin the libray, so I can not use two examples.

How can that been solved?

mitchyboy9, 2017-03-15:
A bit late, but in case it helps anyone else:
Call the `begin` method, passing in the address you want to use.
If you want to use **more** than two MCP4725s in the same project, check out my fork of this project, which allows you to do so.  [This commit](https://github.com/mitchyboy9/Adafruit_MCP4725/commit/7bdeead0e406d226653c62a957dc89762f5a8f80) includes an example of how to do so.

Example showing how to correctly set up addresses using the `begin` method:
```
#include <Adafruit_MCP4725.h>

Adafruit_MCP4725 dac1;
Adafruit_MCP4725 dac2;

void setup(void) {
  dac1.begin(0x62);
  dac2.begin(0x63);
}

void loop(void) {
  Serial.println("Setting 3 different lower voltage outputs");
  dac1.setVoltage(100, false);
  dac2.setVoltage(200, false);

  delay(5000);

  Serial.println("Setting 3 different higer voltage outputs");
  dac1.setVoltage(1000, false);
  dac2.setVoltage(2000, false);

  delay(5000);
}
```

mitchyboy9, 2017-03-17:
I have now written a blog post about this:
http://mitchtronic.blogspot.com/2017/03/addressing-multiple-mcp4724s-in-same.html

ladyada, 2020-06-01:
resolved

closed 2020-06-01 by ladyada